EnviroDIY/SensorModbusMaster #7: need Help

c0zm0z, 2018-09-03:
Hey,
i have used your lib for a long time. I was very happy with it.
Now i am looking for a function for float to CDAB Mid-little endian.

Can you tell me a small hint for adding this function?

SRGDamia1, 2018-09-04:
You would need to add the middle endian type(s) to the endianness typedef and then modify the private "sliceArray" and "leFrameFromFrame" functions and use them appropriately to get the endianness right.

Unfortunately, I will not have time to add this feature myself, but I'm willing to review detailed pull requests.

lthquy, 2021-02-28:
I rewrite function sliceArray to combine float32 from CDAB
It change default bigEndian to CDAB format
`
void modbusMaster::sliceArray(byte inputArray[], byte outputArray[],
                int start_index, int numBytes, bool reverseOrder)
{

    if (reverseOrder)
    {
        // Reverse the order of bytes to get from big-endian to little-endian
        int j = numBytes - 1;
		outputArray[j] = inputArray[start_index + 2];
		outputArray[j-1] = inputArray[start_index + 3];
		outputArray[j-2] = inputArray[start_index + 0];
		outputArray[j-3] = inputArray[start_index + 1];
        /*
		for (int i = 0; i < numBytes; i++)
        {
            outputArray[i] = inputArray[start_index + j];
            j--;
        }
		*/

    }
`

Simonefardella, 2021-12-23:
@SRGDamia1, it's possible to get the raw registers and then do something like that with a union?
```

union {
    uint16_t u[2];
    float f = 0;
} vlnData;

modbus.getRegisters(0x03, 0x5000, 2); //reads two registers starting from 0x5000
vlnData.u[1] = responseBuffer[0]
vlnData.u[0] = responseBuffer[1];

float VoltageL1 = vlnData.f

```

I've tried this approach but i'm not understanding how to access properly the response buffer