EnviroDIY/SensorModbusMaster #8: Spin the Mayfly Wing into its own repo

SRGDamia1, 2018-10-10:
@aufdenkampe I think it should stand on its own, since it's only proximal to the library.

aufdenkampe, 2019-05-22:
I like this idea. Is there a fancy Git way to do this that preserves commit history and issues?

SRGDamia1, 2019-05-22:
Yes, but I'd have to look it up.  You can do things like clone and prune to keep only certain folders.

aufdenkampe, 2021-02-04:
@SRGDamia1, with @neilh10's recent work, it's definitely time to spin the hardware into it's own repo. See https://github.com/EnviroDIY/SensorModbusMaster/issues/14#issuecomment-773421857.

Do you have any suggestions on how to do this in a way to preserve both commits and issues? I'm OK doing the work, including pruning, if I have some guidance. I'll start digging for articles online.

aufdenkampe, 2021-04-01:
@SRGDamia1, I'm going to move forward with this, now that Neil has made additional contributions. I also found these instructions, which seem quite reasonable to follow. https://docs.github.com/en/github/getting-started-with-github/splitting-a-subfolder-out-into-a-new-repository

aufdenkampe, 2021-04-01:
Shall we call the new repo "Mayfly-Modbus-Wing"?

neilh10, 2021-04-02:
Hmm name - valuable for "Presentation"  and searching - ~  how to suggest version 2.

Currently is hardware/Modbus-Mayfly_WingShield  -    and effectively version 1

The board has printed on it 
   KNH002r6 
  Mayfly RS485 

would Mayfly-Wing-Rs485ver2/  for top level 
Mayfly-Wing-Rs485ver2/knh002r6 for specific hardware release.

I personally think the hardware is better represented with RS485, and Modbus is the protocol that sits on top of RS485.

I can see that just being consistent with the Modbus might simplify an early user - so don't have too much opinion on what it is called
so could be  Mayfly-Modbus-Wing2/  for top level 
Mayfly-Modbus-Wing2/knh002r6  for specific hardware package. The numbering "knh002" is somewhat arbitary and just needs to be unique - so I have Kicad NH (for me) and 002 for the board number in my releases, and of course "r6" for the board spin.

However equally could be Mayfly-Modbus-Wing and then the text could I identify it as next generation that supersedes "Modbus-Mayfly_WingShield "

neilh10, 2021-04-02:
So another way of thinking about splitting out SensorModbusMaster/hardware
 https://docs.github.com/en/github/getting-started-with-github/splitting-a-subfolder-out-into-a-new-repository

You'll  create a new repo <Mayfly-Modbus-Wing>  and crate in it current files - readme.MD
https://github.com/EnviroDIY/SensorModbusMaster/tree/master/hardware

Then create a <version1> and put all files <Modbus-Mayfly_WingShield>  changing the text name description to 
"Mayfly-Modbus-Wing version1" - but everything else is same

I could then do <version2> and that would have my specific descriptions including "Mayfly-Modbus-Wing version2"
Then create directory <knh002r6> put all my kicad files into it?

However I can make any of it work .....

aufdenkampe, 2021-04-04:
@neilh10, yeah, that's basically what I was thinking for the process of spinning into a new repo. The new repo would just contain the commits for files presently in the hardware folder. I don't think the issues will follow, but that's OK. We'll then modify the main ReadMe.md, etc.

Regarding naming and releases, my thinking is that the repo name would not include the version number, but we would  handle that via official GitHub releases. We can release historical versions from the commit history (I recently did this for another repo).

For the name, I like either "Mayfly-Modbus-Wing" or Mayfly-RS485-Wing". I recognize that RS485 is the physical communication protocol enabled by the hardware, and that Modbus is a logical communication protocol that is enabled by the software outside of the repo. Nevertheless, for the EnviroDIY community, I might lean slightly in favor of "Mayfly-Modbus-Wing", given it's likely wider recognition and that this hardware is specifically designed to work with the Mayfly to enable Modbus sensors.

What do you all think?

neilh10, 2021-04-05:
Mayfly-Modbus-Wing sounds good, happy with either one.
Well on versions if you want it to be a single stream that can be made to work  - it keeps the view "simple" - the repo keeps the history.  I personally find that I  step into previous releases or want to do a comparison between revisions.  I'll be maintaining my  KNH002r6 sources -  that is the magic number that identifies this  physical board - then  release to <Mayfly-Modbus-Wing> stream as needed.

aufdenkampe, 2021-04-23:
@neilh10, I just made the move! Check it out at https://github.com/EnviroDIY/Mayfly-Modbus-Wing

I just invited you to be a maintainer, so you should have full write+ access. Please share your latest rev7. 

Next steps for me are adding a license and a few other cleanup items.

If things look good to you, I'll create a "historic" release of the original version and start modifying the main readme to describe your new version.

aufdenkampe, 2021-05-21:
All hardware related issues should now be posted at https://github.com/EnviroDIY/Mayfly-Modbus-Wing

closed 2021-05-21 by aufdenkampe