EnviroDIY/SensorModbusMaster #16: Serial to Serial?

mrarmyant, 2020-09-10:
Not using RS485 at all, and compared to another modbus library that was going to force me to write what you have on top of it, im having some issues reading it the same, seems all my bits on an 8n1 setup are shifted 4.  I think its due to some rs485 overhead.  Any suggestions?

SRGDamia1, 2020-10-01:
Sorry for the late response.  Hm.  Well using RS485 or not shouldn't make any difference for this library.  In every Arduino type set-up I've seen, the RS485 to TTL conversion is handled on a separate chip and all the processor sees is the bits in TTL.  So there's either something wrong your whatever-to-TTL chip or there's something wrong with the Stream library you're using to turn the 0s and 1s into characters.  This library doesn't do any of that lower-level stuff.  It needs a stream of correctly read bits to work.

Are you sure that your modbus device actually uses 8N1?  Despite being the really common, 8N1 isn't *technically* modbus compliant.  If your sensor's default is 8O1/8E1, etc, I'd suggest trying to work with that, if possible.  Some sensors just work better in their default configuration.  HardwareSerial is the only Arduino serial I know of that supports anything other than 8N1.  If possible, you should always use HardwareSerial anyway; it's the best option by far.

You said you're not using RS485, but are you still using an half-duplex protocol?  This library does have code in it for using a pin to flip the communication flow direction, but it doesn't work well or at all with a lot of the RS485 to TTL chips.  I've had much better luck with chips or circuits that have automatic flow control.  Modbus is mostly a half-duplex sort-of protocol.  Arduino support for full-duplex depends on the Stream library you're using, but, again, if you're using anything other than HardwareSerial expect it to be error ridden.

aufdenkampe, 2023-10-31:
Closing this issue, as it appears resolved.

If this issue needs further discussion, please continue it in a new issue in https://github.com/EnviroDIY/Mayfly-Modbus-Win

closed 2023-10-31 by aufdenkampe