EnviroDIY/SensorModbusMaster #20: The library does not support esp8266

15670941101, 2021-11-29:
Is there any way to use this library directly on esp8266?

SRGDamia1, 2021-11-29:
I'm not sure.  Have you tried it?  Does it work? If not, what errors do you get?

15670941101, 2021-11-30:
> 我不知道。你试过吗？它有效吗？如果没有，你会得到什么错误？

Yes, I tried, and the result is 0

rishaldyprisly, 2022-01-17:
It still work, i'm using Wemos D1 Mini and D1 Mini Pro, both works fine. But i can't find correct value with int32

rfaass, 2022-03-28:
It owrks for my esp32 lolin32;  there is a debug option that gives you the raw RS485 traffic, that should help in debugging.

hn, 2022-12-05:
You need to update your Arduino ESP8266 board package via the Board Manager to get the latest SoftwareSerial driver. Package V3.0.2 works fine with SensorModbusMaster, V2.6.3 does not.