EnviroDIY/SensorModbusMaster #21: Support Mayfly1.1

neilh10, 2022-02-16:
Support the Mayfly 1.1 +12V_SW.

The Mayfly 1.1 can generate +12V_SW of 12V at 5W of power. 
However the Mayfly 1.1  has NO path power switch. When the boost is disabled, there is still a power path to the input through a diode.

There is a benefit of being able to turn on the boost (typically +12V), allow it to stabilize.
Then when stable the +12V can be applied to the external +12V circuit.

neilh10, 2022-02-16:
A possible solution is attached.
For the power switch MP030D QFN10/0.5mm "USB Charging Port Controller with
Load Detection"

This would use Mayfly Arduino virtual port D9  for  12VswEn

[knh002r8 schematicRs485.pdf](https://github.com/EnviroDIY/SensorModbusMaster/files/8084123/knh002r8.schematicRs485.pdf)

aufdenkampe, 2022-02-22:
@neilh10, thanks for posting this. Since we split off all the hardware stuff to https://github.com/EnviroDIY/Mayfly-Modbus-Wing, let's move the conversation over there.

I just created this issue: https://github.com/EnviroDIY/Mayfly-Modbus-Wing/issues/3

Closing this issue and to move the conversation to that one.

closed 2022-02-22 by aufdenkampe