EnviroDIY/SensorModbusMaster #25: get float 32 from register

Fluxanode, 2022-12-02:
What is the typical usage for 
float modbusMaster::float32FromRegister(byte regType, int regNum, endianness endian)
 {
     getRegisters(regType, regNum, FLOAT32_SIZE/2);
     return float32FromFrame(endian);
 } 
can i call it by: reg1 = modbus.float32FromRegister(0x04, 0x01, bigEndian);?
or do i need to include the rest of the function between the {}?
I'm just a nube learning arduino...

SRGDamia1, 2022-12-06:
You have your endian-ness wrong. This library only works with big-endian (ABCD) and little-endian (DCBA). It doesn't work with mixed-endian numbers. The conversion from 2000447A to 1000.5 is CDAB.

This online converter is very helpful:  https://www.scadacore.com/tools/programming-calculators/online-hex-converter/

Fluxanode, 2022-12-06:
I tried with both big and little.  So are you saying my byte order is wrong?  I'm using ModRSim to simulate and there is no way to change the endian-ness.  I need to be able to talk to a Emerson Micro Motion Coriolis meter.

Fluxanode, 2022-12-06:
Let me check in the meter setup.  I believe I can change the order in the sensor.  I really like your lib by the way!

Fluxanode, 2022-12-06:
FYI - the default for Emerson flow devices is 3-4-1-2 or CDAB.  Can you help me out on changing this in the LIB so there are other options?  It seems that the other sensor types we use (Cameron, Siemens...) are also this format and my bosses want us to be compatible and not require the end user to have to change the configs of the devices.